team401/2024-Robot-Code #11: AprilTag Localization

FujiBookPro, 2024-01-07:
**Description**
We should be able to use cameras to estimate our position via AprilTags.

**Work Scope**
- [ ] Create API/Subsystem to interface with AprilTag code
- [ ] Set up and tune coprocessor running PhotonVision (or other  equivalent software)
- [ ] Wrap over coprocessor in an IO class using NetworkTables
- [ ] Compute pose measurements from NT data

PChild, 2024-01-18:
This can be a new issue if you all want, but can we also get [Photonvision sim](https://docs.photonvision.org/en/latest/docs/simulation/simulation.html) working? 

It would let us test our localization code even when we don't have access to the robot for extended periods of time and can produce video streams like this that would be cool to show judges

[image]

linglejack06, 2024-01-18:
I actually already implemented that with a 2d field packaged in the navigation issue that we pushed to main. All it takes to make field 3d like above is add an extra method call, however, I did not implement it because photon vision states that it takes a lot of processing power to render the 3d field.

jkleiber, 2024-01-18:
@linglejack06 maybe we can make the ticket to implement it, but hide it behind a flag so we can simulate it only when we want to (i.e. when we have the processing power)

linglejack06, 2024-01-18:
yes that makes sense, i will go ahead and add that ticket

jkleiber, 2024-03-02:
This is effectively done software wise, and the main work to do is to tune the kalman filter and make the BeeLink stay powered. Closing

closed 2024-03-02 by jkleiber

redPlover, 2024-03-02:
> This is effectively done software wise, and the main work to do is to tune the kalman filter and make the BeeLink stay powered. Closing

@jkleiber We tugged on one of the power wires for the beelink and it came out. We fixed this so it _could_ fix our issues.