team401/2024-Robot-Code #8: System identification for superstructure

redPlover, 2024-01-07:
**Project Scope**

- [ ] System identification util scalable to all applications
- [ ] Superstructure variation added

jkleiber, 2024-01-23:
We can start by trying the wpilib sysid, and if it meets our needs we can stick with it

redPlover, 2024-01-24:
I've been looking into this recently, I would love to talk about it today at shop, because I think that we should try to implement some of this on our own.

jkleiber, 2024-03-14:
@minhnguyenbhs you could write a program in Python to do the log parsing and system ID (basically an optimization problem to find the system parameters)

jkleiber, 2024-03-14:
To add on, we have an empty sysid repository, which would be the best place to write the system id code from scratch if you take that route