team401/2024-Robot-Code #9: Apply simulation to AdvantageScope

redPlover, 2024-01-07:
**Project Scope**
- [ ] Robot Model Puppeteering
- [ ] Proper that model logging is working

PChild, 2024-01-24:
I started exporting models for this today, it's not 100% since we do have a lot of refinement to do, but I think it'll be close enough that you can start puppeteering and swap in more accurate models once we're done.

redPlover, 2024-01-24:
Okay, I've added the mechanism2d for the scoring subsystem and will send it out as a PR tomorrow.

PChild, 2024-01-24:
GLB files for each system are now posted in the team Google Drive: 
2024 Controls > Main Robot > GLB Files

Based on [the docs](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/CUSTOM-ASSETS.md#3d-robot-models) it sounds like you need zeroed positions and angles, check the OnShape model for those.

redPlover, 2024-02-06:
Done in #64

closed 2024-02-06 by redPlover