team401/2024-Robot-Code #19: Pick vision hardware

jkleiber, 2024-01-11:
**Summary**
We need to pick the hardware for running the vision (AprilTags and Note detection on the ground). 

One thing Preston and I discussed could be:
- 4 black and white cameras for AprilTags (all directions) - these should be in house already
- A high quality black and white camera for front facing AprilTag detection when shooting
- 4 color cameras for note detection
- An additional Beelink
- An amount of Google Corals

See what you think and talk with Preston about it so we can get the hardware ordered

redPlover, 2024-01-12:
I just spoke to him and we are going to test the beelink (once we get it working) with 4 colorless arducams. That list sounds great, but we should probably hold before getting a choral.

jkleiber, 2024-01-12:
Definitely agree on being cautious on the coral, but we will need color cameras for Note detection

redPlover, 2024-01-12:
Sure, I'll try to find some good cameras.

charlesmackenzie, 2024-01-12:
Update: the Beelink boots. It's still not connecting to ethernet so we're reinstalling ubuntu.
[image]

PChild, 2024-01-18:
One of the updates from Photonvision 2024 is support for the accelerator on OrangePis, is it worth getting one to handle color cameras and do note tracking that way instead of just "spot the orange circle"?

jkleiber, 2024-01-18:
@PChild I think it would be totally worth it. At the very worst the RKNN turns out to be a science project that doesn't work well, but we would still be able to get color camera feed.

PChild, 2024-01-21:
Is this complete? Can we document it here or somewhere and then close this out?

jkleiber, 2024-01-21:
@redPlover can you copy paste the rows in that google sheet and then close this? Thanks

redPlover, 2024-01-21:
Sure, here it is:

| Part | Amount | Link |
| ------------- | ------------- | ------------- |
| Orange Pi 5 (8gb) | 2 | [Link](https://www.amazon.com/Orange-Pi-Frequency-Development-Android12/dp/B0BN17PWWB?ref_=ast_sto_dp&th=1&psc=1) |
| Color camera | 4 | [Link](https://www.amazon.com/dp/B00K9ZVB04?ref_=cm_sw_r_cp_ud_dp_HEHKY3MAEVSF2F59JHGT) |
| Beelink | 1 | [Link](https://www.amazon.com/Beelink-4-Cores-Desktop-Computer-Ethernet/dp/B0B17LZ4GD/ref=asc_df_B0B17LZ4GD/?tag=hyprod-20&linkCode=df0&hvadid=647234334956&hvpos=&hvnetw=g&hvrand=12012434723101428199&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1021048&hvtargid=pla-1956454504474&mcid=cc3cb072abfb3c67a7e230ad63377a9d&th=1) |
| Arducam HiRes (50fps) | 2 | [Link](https://www.amazon.com/Arducam-Distortion-Microphones-Computer-Raspberry/dp/B096ZSSCGH/ref=asc_df_B096M5DKY6/?tag=hyprod-20&linkCode=df0&hvadid=647264398501&hvpos=&hvnetw=g&hvrand=1003124240957397689&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9008704&hvtargid=pla-1478895302593&mcid=b3e4b78a31b134b09001072d9e824655&th=1) |

PChild, 2024-01-21:
Can I drop the Pi 5s to 4 GB? I only see one 8 GB model in stock currently. Also any thoughts on powering these?

redPlover, 2024-01-22:
Everyone else seems to use 4 gb, so it's probably fine. I was thinking we splice two usb-c cables (we have a ton from the rev boxes).

PChild, 2024-01-22:
Ok I updated the Amazon order to 2x 4 GB orange pis, added cases for them, and added a pack of 5v 3a buck converters so we can actually power them. 👍

closed 2024-01-22 by PChild