team401/2024-Robot-Code #24: Calibrate Cameras in PhotonVision

jkleiber, 2024-01-14:
**Summary**
Calibration work was underway on 1/13 for cameras on the 2023 robot. This may need to be re-done for future cameras for 2024.

@charlesmackenzie feel free to close this issue if the work is complete or no longer required.

PChild, 2024-01-21:
Can we use [calibdb](https://www.calibdb.net/) to get camera intrinsics instead of the Photonvision calibration? The error comparison in [this report](https://docs.google.com/document/d/1DAPOnU2NfOp91UnMQnkhyiUAanlCDJ6zl_YsCkCMZdA/edit) make it seem like an easy win to improve accuracy

PChild, 2024-02-04:
@charlesmackenzie [Here's the Java camera calibrator](https://github.com/tom131313/calibrate/) we had talked about. Sorry I forgot about it for so long

PChild, 2024-02-07:
Photonvision added [instructions for mrcal](https://docs.photonvision.org/en/latest/docs/calibration/calibration.html#investigating-calibration-data-with-mrcal) which is supposed to be even better.

FujiBookPro, 2024-02-11:
Calibration data from calibdb for all of our new cameras has been uploaded to Google Drive

closed 2024-02-11 by FujiBookPro