team401/2024-Robot-Code #25: Get a simple autonomous working

jkleiber, 2024-01-14:
**Summary**
To establish a foundation for future work in autonomous / drivetrain, we should get a simple auto drive working using PathPlanner. This was started by @minhnguyenbhs on 1/13, and this issue will track this work.

**Work Scope**
- Using path planner and the default path following stuff we worked with in the offseason, iron out bugs for auto driving simple paths.
- Don't focus on tuning, but rather on getting the robot to move the right direction and stop at the correct events
- Use the following incremental buildup (don't move on until satisfied with robot performance):

- [ ] Move straight backward
- [ ] 2 notes 
- [ ] 3 notes (assume the intake is in the back)
- [ ] 4 notes
- [ ] 5 notes

It would be good to get this working in simulation, but this depends on #6. Since we have a physical robot at the moment, let's focus on using that for now.

redPlover, 2024-01-14:
I'm currently working on getting sim working for @minhnguyenbhs to use. I thought that it'd be simple with just logging the preexisting one in AK but I'm having some lag issues. I should be done by tonight.

redPlover, 2024-01-28:
@jkleiber
The auto branch was outdated enough to where it was easier to create a new one off of main and copy & paste @minhnguyenbhs's code. It is now in auto-modern. I added a sample auto path with NamedCommands that aims the robot and shoots.

jkleiber, 2024-02-07:
@redPlover assigning this to you - we really need this to be driven to closure so we at least can do baseline auto testing in sim prior to getting the robot

jkleiber, 2024-02-07:
This is done now that #65 merged. Closing.

closed 2024-02-07 by jkleiber