team401/2024-Robot-Code #34: Document Control Flows with Mermaid

PChild, 2024-01-18:
Having control flow diagrams in the readme is really cool, could we use Mermaid to make them interactive / a little fancier instead of just screenshots? 

Here are a couple of references: 
- [Github docs link](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-diagrams)
- [Mermaid docs](http://mermaid.js.org/syntax/stateDiagram.html)

And an example of a simple flow chart:

```mermaid
graph TD;
    A-->B;
    A-->C;
    B-->D;
    C-->D;
```

redPlover, 2024-01-18:
Can we transfer draw.io diagrams to mermaid?

jkleiber, 2024-02-01:
@aidnem we can use this ticket for any documentation beyond the shooter

aidnem, 2024-02-01:
@jkleiber Quick question about the preferred style for a logical or (`||`) to be drawn in. There are two options (as far as I can see):
1. Displaying it in parallel (the previous step leads to both or conditions, which would lead to the 'yes'/'true' condition from both of them or back to the original state for the 'no'/'false') condition.
2. Display it sequentially: The starting state leads to the first condition. The first condition, if true, leads to the true result. If false, the first condition leads to the second condition, which, if true, leads to the true result as well. If false, the second condition leads back to the original state.
I am currently leaning toward option 2 (sequential), as it seems to more accurately represent the actual flow of the code.
Both examples are demonstrated below:
```mermaid
flowchart LR

subgraph parallel
someState
or1
or2
orResult
someState --> or1
someState --> or2
or1{some_variable == SOME_VALUE}
or1 -->|yes| orResult
or1 -->|no| someState
or2{something_else >= 2}
or2 -->|yes| orResult
or2 -->|no| someState
orResult[Result of 'or' operation is true]
end

subgraph sequential
someState_
or1_
or2_
orResult_
someState_ --> or1_
or1_{some_variable == SOME_VALUE}
or1_ -->|yes| orResult_
or1_ -->|no| or2_
or2_{something_else >= 2}
or2_ -->|yes| orResult_
or2_ -->|no| someState_
orResult_[Result of 'or' operation is true]
end
```

jkleiber, 2024-02-12:
I'm going to leave this open for now so we can fix the small defects in the automatic text generated in files like https://github.com/team401/2024-Robot-Code/blob/main/doc/Scoring.md