team401/2024-Robot-Code #40: Always align drivetrain to various targets

jkleiber, 2024-01-19:
**Purpose**
The purpose of this issue is to add the ability to be always aligned with the goal. This will require an additional layer of control laws that interfaces with the driver's commands.

Places we want to align:
- Speaker
- Amp
- Source

We may also want to align with the stage automatically, but that can be added later.

**Concept of Operations**
The driver will switch to this mode via a button / trigger (which will be associated with a state in the Drivetrain state machine). They will be locked out of the ability to rotate the drivetrain. Instead, they will issue field relative translation commands, and the robot will automatically align to a target chosen by the driver / button masher via button press / POV on the gamepad (this will need to be indicated on the dashboard / copper console).

The Drivetrain will rotate to align with targets using a PID controller (auto-aim). If left unchecked, the rotational motion of this controller will cause a bias that pushes the robot away from the driver's commanded translation. Therefore, additional control loops may need to be designed to take the human commands and correct for the drift induced by the auto-aim controller.

Off the top of my head, the additional control loops could just lookup a "restoring force" based on the auto-aim controller's `omega` output. A more robust solution would be a PID that takes ground track error (i.e. human controls 90 deg but robot is moving at 95 degrees = 5 degrees of error), and then that outputs a correction factor to add onto the joystick velocity commands. This second approach would need to only activate at high enough speeds to avoid weird behavior (at slow speeds the robot will not have a reliable ground track, which could result in fictitiously large ground track error and overcontrol).

**Project Scope**
- [ ] Add state for auto-aim to the drivetrain state machine
- [ ] Add PID controller for auto aiming
- [ ] Create setpoint calculation logic for auto-aim (some targets are position based, so the rotation will change based on robot position, while others are pure headings. The selection of a target will dictate the yaw setpoint we pass to the PID controller).
- [ ] Create a control system that can make the translation control work as a human driver expects (eliminate drift)

aidnem, 2024-02-06:
Still needs more work to eliminate drift

jkleiber, 2024-02-25:
Moving off the VABLA integration campaign milestone since the remaining work is more complicated and lower priority

jkleiber, 2024-03-05:
I'm going to close this issue in favor of more specific issues based on driver feedback. I think the main feature is done and the drift is something we will compensate for when improving driver UX if the driver wants that.

closed 2024-03-05 by jkleiber