team401/2024-Robot-Code #45: Document scoring subsystem and its state management

redPlover, 2024-01-21:
**Purpose**
The purpose of this addition is to document how our state control works in the scoring subsystem.

**Project Scope**
- [ ] Add state transfer matrix
- [ ] Implement block diagram for all states (_amp & endgame will be added later_)
    - [ ] IDLE
    - [ ] INTAKE
    - [ ] PRIME
    - [ ] SHOOT

redPlover, 2024-01-22:
This should use #34

jkleiber, 2024-01-29:
@aidnem assigning this to you - will want to talk about this more at tonight's meeting

aidnem, 2024-01-29:
@jkleiber I won't be at tonight's meeting due to a carpool situation, I should be there Tuesday, Wednesday, and Saturday this week if any of those work. If needed I can arrange to come on future Mondays if necessary.

redPlover, 2024-01-29:
@aidnem I can give you ride

redPlover, 2024-01-29:
@aidnem I can give you a ride

aidnem, 2024-01-29:
@redPlover I appreciate the offer, but my family made other plans. If work is needed urgently on this, you all can do whatever's necessary (I'd probably be heavily using your advice anyway) and I can update Copper Console and anything else to fit whatever scheme you decide (or I can weigh in later in person or through GitHub Issues).

jkleiber, 2024-01-29:
@aidnem while the controls team meets every day, I am only there on Mondays, Wednesdays and Saturdays. We can sync on Wednesday.

If it isn't very inconvenient it would be preferable to come in on Mondays rather than Tuesdays, but I understand if that's not possible

closed 2024-02-07 by jkleiber