team401/2024-Robot-Code #47: Create test plan for shooter speaker tuning

jkleiber, 2024-01-23:
**Purpose**
The purpose of this issue is to develop a test plan for the shooter. The goal of the test plan is to determine what angles and RPMs are needed to make shots at variable distances from the speaker

**Project Scope**
- [ ] Develop test plan for shooting into the speaker
- [ ] Add test plan as a markdown file in the repo, and link it to main shooter documentation

jkleiber, 2024-02-10:
@redPlover @charlesmackenzie I've started a document here: https://docs.google.com/document/d/1Q0RYd344qq8yRoueZuRTNFPpY1EmJdiOKtMOQQR6ShU/edit

I focused on making the subsystem verification plan but we need to write down how we are going to do the shooter test

redPlover, 2024-02-12:
I'll start working on the "tune mode" for testing

jkleiber, 2024-02-21:
This is effectively done by the work in the scoring integration branch. Any future work can be captured in separate issues

closed 2024-02-21 by jkleiber