team401/2024-Robot-Code #48: Planning for LED signalling

PChild, 2024-01-23:
**Purpose**
The purpose of this addition is to plan for LED signalling

Initial thoughts on LED sequencing:
- off for not connected
- rainbow for disabled and connected
- white for intaking, flash on acquire note
- green for ready to shoot? (wheels at speed, angle good, vision lock is good?)
- some color for "has note"?
- flash yellow or something for last 30 seconds

**Project Scope**

- [ ] Add an LED subsystem
- [ ] Establish an API
- [ ] Implement LED code
- [ ] Simulate it (within reason)

charlesmackenzie, 2024-02-07:
The basic patterns should look as follows:
- Intaking: orange
    - Briefly flash when we get a note
- Speaker scoring: alliance color
- Amp scoring: green
- Endgame: purple
- Default (enabled): fun animation (undecided)
- Disabled (FMS connected): rainbow animation
- Disabled (disconnected): No lights

jkleiber, 2024-02-13:
#95 implemented LEDs, with some level of smartdashboard simulation. I think we can close this issue and do any follow on work for LEDs as needed

closed 2024-02-13 by jkleiber