team401/2024-Robot-Code #52: Test Disabling RoboRIO Web Server

PChild, 2024-01-24:
**Purpose**
AdvantageKit can be more memory intensive than a stock wpilib project, disabling the RIO web server can buy us a bit more RAM back. A recent [wpilib update ](https://github.com/wpilibsuite/allwpilib/pull/6270) added the ability to disable and reenable the webserver through a GUI. It would be good to compare RIO free memory running our current code, after disabling the web server, and after replacing lvrt.

Details on replacing lvrt are available in [this ChiefDelphi post](https://www.chiefdelphi.com/t/fatal-robot-code-crash/427074/44)

**Project Scope**
- [ ] Update a roboRIO / all of our roboRIOs to 2024
- [ ] Record free memory when running our 2024 code
- [ ] Disable RIO web server, rerecord free memory
- [ ] Check for persistence of disabling the server
- [ ] Check that re-enabling the server works correctly
- [ ] Test replacing lvrt on the RIO, rerecord free memory

redPlover, 2024-02-06:
## Here are the performance improvements of disabling the webserver and lvrt. All of these are with the drivetrain code running.

### Normal
[image]
### Webserver Disabled
[image]
### Webserver & LVRT Disabled
[image]

jkleiber, 2024-02-06:
What exactly are we looking for in these pictures?

PChild, 2024-02-07:
Our rough testing showed that the (drive only) robot program was using ~10 MB of RAM. I'd like to check later in the season when we have more subsystems enabled and see how memory use compares.

closed 2024-02-07 by PChild