team401/2024-Robot-Code #58: Add shoot on the move

jkleiber, 2024-01-27:
**Purpose**
The purpose of this addition is to support shooting notes while driving.

**Derivation**
[image]

see the above picture. First we need to compute an anticipation distance based on the robot speed. Then we figure out what base direction / aimer angle / shoot power is needed at that anticipation along the robot's velocity (the right part of the image). Third, we compute an adjustment for the robot's velocity and shooter RPM such that the Note doesn't get "pushed" along by its initial velocity.

The following considerations need to be made:
- Robot angle tolerance
- RPM tolerance
- How we generate the lookup table for RPMs and shooter angles

**Project Scope**
- [ ] Add anticipation distance to shoot target calculations
- [ ] compute shoot target parameters based on arbitrary x,y coordinates rather than robot position
- [ ] Add correction factors for robot velocity
- [ ] come up with a general idea for how to generate the lookup table for RPM and shooter angle (may be another ticket prior to working this one tbh)

redPlover, 2024-01-27:
@jkleiber Do you mind if I take this one? I'm interested in figuring it out.

jkleiber, 2024-01-28:
Go for it @redPlover ! Although it's not the highest priority rn obviously

redPlover, 2024-01-28:
@jkleiber Should I contribute to auto instead?

jkleiber, 2024-01-28:
@redPlover yeah let's at least get a first draft of auto in Path Planner working with the software in sim before we do this

We'll also need to figure out how to fix the sim thing you showed me today - I'll see if it happens on my computer today or tomorrow

charlesmackenzie, 2024-01-30:
Here's a [ChiefDelphi](https://www.chiefdelphi.com/t/shooting-while-driving-good-shooting-code/) thread about shooting while moving for this year. It links to [this visualization](https://www.desmos.com/calculator/ejg6jrsodq) on Desmos.

I should note that the solutions being proposed in the thread take the opposite approach to @jkleiber's diagram-- they think about velocities from the reference frame of the robot, whereas we're addressing the field reference frame

closed 2024-02-03 by redPlover