team401/2024-Robot-Code #70: Adjust RPM during shoot on the move

jkleiber, 2024-02-03:
**Purpose**
The purpose of this addition is to adjust the shooter RPM based on the robot's velocity while shooting on the move.

This was not originally added in #67 to avoid additional complexity when making dynamic adjustments to the anticipated setpoints. This ticket will only be worked if it is determined that we need it for normal shoot on the move operations.

**Note:** The MS paint diagram on the left of #58 explains a possible way to solve for the rotation and RPM adjustment at the same time.

**Project Scope**
- [ ] Add an adjustment factor for the shooter RPM based on robot field relative x velocity
- [ ] Update the rotation adjustment according to how the RPM is adjusted. In reality, these will need to be adjusted together.
- [ ] Update docs to explain how the implementation works

aidnem, 2024-02-15:
@jkleiber velocity to goal calculation is working well. I'm wondering how to translate this to a change in the RPM of the motor. Currently, the motor rpm is set by a mapping of distance to RPM:
```java
shooterIo.setShooterVelocityRPM(shooterInterpolated.getValue(distancetoGoal));
```
Should I make the velocity subtract from this RPM? If so, should I make it linear, or should I use another lookup table, as such?
```java
shooterIo.setShooterVelocityRPM(shooterInterpolated.getValue(distanceToGoal) - velocityToRpmInterpolated.getValue(velocityToGoal);
```
Alternatively, we could map distance to goal to a desired velocity, then subtract the robot's velocity to goal from this, and then use this new desired velocity with another mapping from velocity to RPM (This is the cleanest option but it would be a massive pain to tune).

jkleiber, 2024-02-15:
@aidnem let's make a lookup table like you suggested so we can have more control over the tuning.