team401/2024-Robot-Code #73: Simulate note

redPlover, 2024-02-03:
**Purpose**
It would be beneficial if we could simulate the note's trajectory and show it moving in AdvantageScope. This entails adding a `NoteSubsystem.java`.

**Benefits**
- [ ] Better pre-tuning
- [ ] Cool logging
- [ ] Good judge material
- [ ] Preston is happy

PChild, 2024-02-09:
[Related](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/tabs/3D-FIELD.md#game-piece-objects) 😃

redPlover, 2024-02-09:
> [Related](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/docs/tabs/3D-FIELD.md#game-piece-objects) 😃

Nice! I'll look into this once MV is done

jkleiber, 2024-02-13:
@redPlover I'm assigning @minhnguyenbhs to this as well since this is longer term simulation work than what you're currently working on. It would be good if you could review the work she does and talk to her about ideas you have.  No action needed from you now: I just think this may be valuable pre-Blacksburg and I would like to keep the progress going in parallel to the critical path

minhnguyenbhs, 2024-02-22:
sorry this is my personal list of todo things for next shop

- [ ] figure out how to get the actual robot angle
- [ ] after real interpolate values are found make sure calculated velocity is actually correct
- [ ] work on figuring out when note travels through the speaker (work on 3D PIP?)
- [ ] note intake
- [ ] maybe get a proper cad note?

PChild, 2024-02-22:
@minhnguyenbhs I exported a glb file of the Note CAD for you, [available here](https://drive.google.com/file/d/1LnZhZMsx9ZjxUuOLo1UfP_EiuvPWc8Lx/view?usp=sharing).

closed 2024-04-15 by minhnguyenbhs