team401/2024-Robot-Code #81: Consider acceleration when determining shoot on the move anticipation distance

jkleiber, 2024-02-05:
**Purpose**
The purpose of this addition is to improve the fidelity of our anticipation distance estimate by incorporating the robot's current acceleration (or deceleration) into the calculation.

For example, rather than computing `x = vt` we can compute `x = vt + 0.5*a*t*t` if the acceleration is sufficiently filtered.

This is a fairly low priority until we can actually test shoot on the move and identify any deficiencies.

**Project Scope**
- [ ] Add a filtered acceleration to telemetry
- [ ] Incorporate the acceleration into the anticipation distance calculation

linglejack06, 2024-02-22:
@jkleiber would i add the acceleration calculation to the robot velocity correction angle in calculate desired heading method?

jkleiber, 2024-02-22:
@linglejack06 yeah. The way it works is we predict our future position based on velocity, so you'd add in acceleration to improve the estimate

Currently we have 
x = vt

But we would want
x = vt + 0.5*a*t*t 
For higher accuracy

linglejack06, 2024-02-25:
wontfix

closed 2024-02-25 by linglejack06