team401/2024-Robot-Code #85: Configure Control Scheme With JSON Files

charlesmackenzie, 2024-02-06:
**Purpose**
Manually configuring robot controls in `RobotContainer` is often cumbersome, and doesn't easily allow for multiple controller configurations (eg: single gamepad, two sticks + gamepad). It would be beneficial to save controller configurations in our deploy folder, and select one by setting a variable in constants.

**Project Scope**
- [ ] Find/make a JSON setup to define controllers and button mappings
- [ ] Create a class to read these files and generate triggers from button mappings
- [ ] Add a menu in CopperConsole to create/edit controller configs (?)

**Optional**
We aren't likely to need to change controller configurations at runtime, so we could feasibly extend the build system to generate a human input class from a JSON at buildtime, instead of deploying all our configurations to the robot and reading one at init. I should make it clear I'm not saying this is a good use of our time.

charlesmackenzie, 2024-02-07:
WPIlib seems to use [FasterXML/jackson](https://github.com/FasterXML/jackson) to parse JSON files.

jkleiber, 2024-02-26:
This is not looking likely to go in prior to Blacksburg, so I am moving this to the Falls Church milestone