team401/2024-Robot-Code #86: Investigate using git hooks to avoid accidentally failing CI

charlesmackenzie, 2024-02-06:
**Synopsis**
It seems that, very often, someone will make last-minute changes to a PR that cause it to fail the "format" action. This might be mitigated by automatically running `gradlew spotlessJavaCheck` on the `pre-commit` git hook to ensure the commit will pass the formatter before it's pushed.

**Pros**
`spotlessJavaCheck` runs much faster locally than it does on Github Actions, so >30s of time is saved per PR, along with any  extraneous commits or force pushes.

**Cons**
Our project workflow may become more cumbersome to explain to rookies, and depending on obscure git features almost never represents a stability improvement.

jkleiber, 2024-02-06:
This could be a good idea depending on the complexity of setup and the configurability/visibility of the change.

It would be good work on the habit of building before pushing though. For our project, CI exists to protect main.

FujiBookPro, 2024-02-23:
The short answer, after some searching, is no.

It doesn't seem like git tracks hook scripts by default, so the most common approach is to use a tool called [pre-commit](https://pre-commit.com/).

I don't think it's a good idea to depend on an obscure, Python-based CLI tool for our robot project, so I'm closing this one for now.

closed 2024-02-23 by FujiBookPro