team401/2024-Robot-Code #98: Use a PID + Motion Profile to control the climber

jkleiber, 2024-02-08:
**Purpose**
The purpose of this addition is to upgrade the baseline climbing controls to use a PID and a trapezoidal motion profile.

The current baseline is to run the climber into the hard stops at some constant voltage and then go into brake mode with 0 voltage. The safety feature to turn off the climber at the current limit should be kept, but it would be an improvement to control to some target.

We will have the motor's internal encoder available - while not precise, this should be sufficient for this application.

**Project Scope**
- [ ] Add PID and trapezoidal profile to the endgame subsystem
- [ ] Add encoder targets for up and down positions
- [ ] Demonstrate that this works in simulation

redPlover, 2024-02-08:
@jkleiber was thinking about that earlier so I already started writing this in the "integration branch". I can roll back though and only add the encoder in? I definitely think that Aiden should work on this.

jkleiber, 2024-02-08:
Yeah let's just get something testable (and easily verifiable in sim) abd then use this as a good first controls task

jkleiber, 2024-02-13:
@aidnem this is something that is good to work on when you are tired of working on RPM adjustment

aidnem, 2024-02-18:
@jkleiber Are we trying to integrate this tomorrow at shop? I'm wondering how much I should prioritize getting it done (as much as I know how) today, whether it could be pushed back a little, and if I should come to the shop on Monday or Tuesday.

jkleiber, 2024-02-18:
@aidnem if we have it available to test by tomorrow then we may try it out.

Take a look at the integration branch to see what updates @redPlover needed to make to the endgame IO - it may be worth it to base your branch off of that branch anyway so it can go into test quicker

aidnem, 2024-02-18:
@jkleiber OK, I'll try to have it done, and I'll see if I can come to shop on Monday.

Should I just merge the [scoring-real-integration](https://github.com/team401/2024-Robot-Code/compare/scoring-real-integration) branch into my feature branch and then continue working?

jkleiber, 2024-02-18:
Yeah that sounds good

jkleiber, 2024-02-29:
This is no longer in scope for Blacksburg, mainly because we can do the climb manually fairly easily provided we can see what we are doing. This will be needed before Falls Church

closed 2024-03-06 by aidnem