team401/2024-Robot-Code #99: On-the-fly trajectories and following to field locations

PChild, 2024-02-08:
**Purpose**
The purpose of this addition is to add on-the-fly trajectory generation and following to the robot project. This should allow us to automate most if not all scoring actions when undefended.

**Project Scope**
- [ ] Use [PathPlanner OTF](https://pathplanner.dev/pplib-create-a-path-on-the-fly.html) to generate trajectories from current pose to important field locations (speaker? amp, source)
- [ ] Implement controls to allow drivers to use this
- [ ] Test in sim
- [ ] Test on hardware

PChild, 2024-02-08:
I set this as Advanced because I'm not sure we want to prioritize it before Blacksburg, but it would be nice to have :)

redPlover, 2024-02-08:
I started some of the boilerplate necessary for this in the auto PR. It "should be fairly easy" to incorporate.

jkleiber, 2024-02-09:
Yeah this is firmly in the Advanced category (i.e. we should prioritize all MVP work over this)

jkleiber, 2024-03-05:
This likely will be needed for #123

jkleiber, 2024-03-14:
Arbitrary OTF will likely not make it in for VAFAL, moving to DCMP

linglejack06, 2024-03-14:
I saw honzik added a OTF path generation command for climb, is it a lot of extra code to just add different poses in there? If not I could code that this weekend to have it ready.

jkleiber, 2024-03-14:
@linglejack06 it's totally worth it for us to have additional locations if possible. We'll just need to get our climb stuff in first

jkleiber, 2024-04-15:
@aidnem / @linglejack06 FYI, I'd like us to start making progress on this ticket in support of #182. Ideally they could be worked in parallel