team401/2024-Robot-Code #142: Gracefully handle arm rotation motor failure

PChild, 2024-03-04:
**Purpose**
The purpose of this addition is to handle the eventuality when we may only have one motor driving arm rotation. We could have an emergency mode (toggled on the dashboard maybe?) that switches our gains.

**Project Scope**
- [ ] Tune gains for single arm motor case
- [ ] Add control on dashboard to toggle gain sets
- [ ] Test it

jkleiber, 2024-03-05:
What are the failure modes for this system? If we understand them, we could probably automatically detect motor failure and switch to backup gains.

For example, if one motor is not drawing (much) current relative to the other motor, we could assume that that motor is dead/disconnected/fell of the sub-assembly. We could also likely detect CAN errors for a motor and handle those similarly

jkleiber, 2024-03-14:
I want to get this in prior to VAFAL if possible, but it is not strictly required. Moving to DCMP milestone