team401/2024-Robot-Code #149: Integrate note sensor in uptake and prevent multiple note intake

jkleiber, 2024-03-04:
**Purpose**
The purpose of this addition is to add note sensor support in the uptake so we don't get a ton of technical fouls

**Project Scope**
- [ ] Add support for a sensor to detect notes in the uptake
- [ ] Include this sensor in logic for the intake to prevent the robot from picking up multiple notes
  - [ ] If a note is in the uptake and shooter deck, run the uptake and intake backwards to get rid of the note
  - [ ] If a note is in the uptake, run the intake motor(s) that touch notes on the ground backwards

jkleiber, 2024-03-19:
While this work is basically done, we will not have time to test it prior to Falls Church. Moving to the DCMP milestone

PChild, 2024-03-26:
Moving this to DCMP milestone, doing some cycle analysis we're losing time when waiting to know that we have a note, this sensor would let us change LED state earlier and save time every cycle.

closed 2024-04-03 by redPlover