team401/2024-Robot-Code #150: Add even better autos for VAFAL

jkleiber, 2024-03-04:
**Purpose**
The purpose of this addition is to make better, faster, and higher scoring autos for Falls Church.

**Auto Options**
- [ ] Pure 6 note (need to be able to drive fast, vision support)
- [ ] 2 notes from centerline
- [ ] 3 notes from centerline
- [ ] Any 4 notes off the floor from any start position

aidnem, 2024-03-13:
@jkleiber What did you mean by "Any 4notes off the floor from any start position"? Should I just make a few 5-note autos (4 off the ground + 1 starting note) to have as options, or did you mean something else?

jkleiber, 2024-03-14:
@aidnem Basically if we can add useful/complimentary 5 note autos from each start position that is the goal

jkleiber, 2024-03-20:
Reopening to track auto work still in progress prior to the event

closed 2024-03-25 by jkleiber