team401/2024-Robot-Code #171: Pay off AutoBuilder Tech Debt

jkleiber, 2024-03-25:
**Purpose**
The purpose of this issue is to use AutoBuilder correctly so we can use the waypoints in the individual paths

**Project Scope**
- [ ] Follow the AutoBuilder docs for creating autos
- [ ] Test in sim that the code doesn't crash. 
- [ ] If #170 is available, test driving in sim, else try on the robot during auto testing

redPlover, 2024-03-27:
@jkleiber I found the issue. We were creating the named commands in the wrong sequence. They work now and our sneaky 4 note worked first try afterwards. Unfortunately that was the only time it worked tonight. I'll keep investigating, but I tuned drive afterwards and it seems to track better.

jkleiber, 2024-03-27:
@redPlover that's great news! Are we missing shots or are we running into other problems?

redPlover, 2024-03-27:
> @redPlover that's great news! Are we missing shots or are we running into other problems? 

@jkleiber We're running into other problems, mostly with drive. All the shots it wanted to make made it, sometimes it didn't want to shoot though. I tried override shooting at the end of every X-E1 path to see if that'd help but it caused us to miss. I think that most of the issues will be fixed once we get the drive controller working well.

closed 2024-04-03 by redPlover