team401/2024-Robot-Code #190: Make single tag navigation viable

jkleiber, 2024-04-15:
**Purpose**
See title. If we can get single apriltag navigation working then it will be very good for next season and the offseason

**Project Scope**
- [ ] Figure out how to get single tag nav working

Please post progress notes in this ticket's comments

minhnguyenbhs, 2024-12-08:
hi! is anyone working on this right now?
I'm almost done with the cad file stuff- I got the necessary files from preston and read about how to fix the branch yesterday so it hopefully shouldn't take too long to finalize that branch on monday- so I can do something else while elevator gets finalized :D

jkleiber, 2024-12-08:
@minhnguyenbhs go for it! Looks like you've already assigned yourself so you can work on this 👍