team401/2024-Robot-Code #202: Add coppercore using Jitpack

aidnem, 2024-08-19:
**Purpose**
Add coppercore as a dependency using [jitpack](https://jitpack.io/). Potential alternative to #201 .

**Project Scope**
- [ ] Add jitpack to build script
- [ ] Use coppercore using jitpack
- [ ] Use coppercore features

aidnem, 2024-08-20:
@jkleiber it seems I'm getting somewhere! Poking around in some of the links in the error messages brought me here: [the build log from jitpack's attempt to build our library](https://jitpack.io/com/github/team401/coppercore/json-controller-incorporation/build.log). At the bottom, it states "⚠️ ERROR: No build artifacts found". I assume this means that if we can make our library produce an artifact(s?), it will let us build it.

jkleiber, 2024-08-20:
@aidnem have you tried looking at this: https://jitpack.io/docs/BUILDING/
we might need to add something on the coppercore side to get it to build correctly in jitpack

aidnem, 2024-08-20:
@jkleiber looks like I have to enable building publications in the build.gradle. Shall I branch off of Minh's JSON Controller Implementation branch just to give me something to import?

aidnem, 2024-08-20:
I found a post online linking to an example project, these commits are what we're looking for it seems:
[1: Using the published modules](https://github.com/vestrel00/jitpack-publish-sample/commit/4661ec2e79b90220945b6e0c7a020d3c3903adb8#diff-db65d28979fd268d1c365b281880e02a47354a9c17c6767d5d0d50e899aa0b71)
[2: Enabling publishing of the modules](https://github.com/vestrel00/jitpack-publish-sample/commit/d8bc5cdcba8b48db9d5797ed7740e03967fb9b76#diff-af0a526d0e3d30a4f30d3ce9d903a61f894850626d8161c6ed941d9ce4b0c641)